rusefi/uaefi #2: fun ticket: STM32F407VGT6 price and availability

rusefillc, 2023-09-11:
https://jlcpcb.com/partdetail/Stmicroelectronics-STM32F407VGT6/C12345

[image]

rusefillc, 2023-09-11:
See also https://github.com/andreika-git/hellen-one/issues/19

rusefillc, 2023-09-16:
What's going on? stm32 cheaper than Geehy?!

https://jlcpcb.com/partdetail/Geehy-APM32F407VGT6/C2990873

well yes stock goes down let me go grab some more :)

[image]

ElDominio, 2023-09-16:
[image]

After 30pc it IS cheaper

Worth it is a different story

rusefillc, 2023-09-21:
[image]

rusefillc, 2023-09-23:
[image]

rusefillc, 2023-10-07:
they got more!

[image]

rusefillc, 2023-11-29:
[image]

rusefillc, 2023-11-30:
Looks like $4 is the official reality these days

closed 2023-11-30 by rusefillc

rusefillc, 2024-04-11:
[image]