rusefi/uaefi #35: cost reduction ideas

rusefillc, 2024-01-13:
J2/J3/J4/J5 alternatives same as J10

rusefillc, 2024-01-13:
@Qwerty-OFF do we care to use specifically C32677 for "19V 13.3V 12V SOT-23(TO-236) TVS ROHS"

оно basic но относительно дорогой. extended бывают сильно дешевле https://jlcpcb.com/partdetail/Onsemi-MMBZ20VALT1G%2FC153067#EC известный бренд, это подходит?

rusefillc, 2024-01-13:
@Qwerty-OFF усилитель это важно, мы никому тут не верим?

[image]

Qwerty-OFF, 2024-01-13:
PSM712 можно поменять на другой extended дешёвый PSM712. Твой вариант не подходит

Qwerty-OFF, 2024-01-13:
Усилитель китайский китай брать не стоит, возможно искать какие-то аналоги от нормальных производителей дешевле

rusefillc, 2024-01-28:
> Усилитель китайский китай брать не стоит, возможно искать какие-то аналоги от нормальных производителей дешевле

ну или https://github.com/rusefi/uaefi/issues/40

closed 2024-01-28 by rusefillc