rusefi/uaefi #36: physically larger pulldown 4.7k-10k on button lines

rusefillc, 2024-01-15:
at the moment button lines pull up/down is done using 0402 which is challenging to adjust manually

we shall add 0603 or better 0805 footprints for button inputs

UAEFI сейчас имеет три входа "button" pullup 10k. Заменяем pullup на 680k, это не сломает работы с обычными кнопками замыкающимися на землю. В адаптере на линиях "button" ставим pulldown 4.7k-10k, за счет этого можно будет использовать данные линии для датчиков/кнопок коммутирующих 12В.

Qwerty-OFF, 2024-01-26:
Какие установить подтяжки в конфигурации по умолчанию?

rusefillc, 2024-01-26:
10k PU 0805
680k PD 0402 на MM?

Qwerty-OFF, 2024-01-28:
[image]
Сделал 10k 0805 PU в этом поле и 680k 0402 PD внутри MM

Qwerty-OFF, 2024-01-30:
Кажется мы запутались, и нам нужно PU внутри ММ для кнопок замыкающихся на землю, и сильный PD снаружи для более странных кнопок

Qwerty-OFF, 2024-02-05:
Оставляю по умолчанию все button-линии с 0805 pulldown 10k пригодным для использования с кнопками замыкающимися на 12В.

closed 2024-02-05 by Qwerty-OFF