rusefi/uaefi #37: подписать рекомендованные назначения пинов синим

rusefillc, 2024-01-19:
не меняя названия сетей на схеме хочется где-то голубым добавить комментарий про рекомендации

Fan Relay https://rusefi.com/docs/pinouts/hellen/uaefi/?connector=b&pin=B8 
Main Relay https://rusefi.com/docs/pinouts/hellen/uaefi/?connector=b&pin=B9
Fuel Pump Relay https://rusefi.com/docs/pinouts/hellen/uaefi/?connector=b&pin=B16
Idle Air Valve https://rusefi.com/docs/pinouts/hellen/uaefi/?connector=b&pin=B18

примерно вот так
[image]

rusefillc, 2024-01-19:
вторая половина https://github.com/rusefi/kicad6-libraries/issues/15

rusefillc, 2024-01-19:
@Ringenium 
VVT1 https://rusefi.com/docs/pinouts/hellen/uaefi/?connector=b&pin=B7
CAM1 https://rusefi.com/docs/pinouts/hellen/uaefi/?connector=c&pin=C5
CAM2 https://rusefi.com/docs/pinouts/hellen/uaefi/?connector=c&pin=C6

Ringenium, 2024-01-19:
Все добавил в компонент

closed 2024-01-19 by rusefillc