rusefi/uaefi #54: make on-board MAP sensor pinout more obvious

rusefillc, 2024-03-03:
a talented individual can solder mpx4250 more than one way, we need some clarification note

[image]

ggurov, 2024-03-03:
[image]

rusefillc, 2024-03-03:
I am tired and slow

Just realized that we have a drawing of the vacuum line outlet which tells how to position, not sure if any clarification is needed on top of that.

TIL about notch on the first pin!

ggurov, 2024-03-04:
should mark the input pin. the drawing of the vacuum line did not help. i went for the multimeter to and started tracing stuff. maybe put PC1 next to that through hole for the notch, or  put "notch" next to it as well.

Qwerty-OFF, 2024-03-27:
Added pin numbers
[image]

closed 2024-03-27 by Qwerty-OFF