rusefi/uaefi #57: pad for resistor between VR lines

rusefillc, 2024-03-18:
* Do we have a pad to mount a resistor between both VR lines? 
* is it easy to use?
* do we have a note on the schematics explaining, similar to Frankenso?

rusefillc, 2024-03-23:
@Qwerty-OFF do we already have similar resistor between lines anywhere?

rusefillc, 2024-03-23:
max9924 already has it just needs documentation https://github.com/andreika-git/hellen-one/issues/374

Qwerty-OFF, 2024-03-27:
[image]

https://github.com/andreika-git/hellen-one/commit/21de8c2cf88dabbff60d48988fd028638cbae64e
https://github.com/andreika-git/hellen-one/commit/48b12a8b670ce0aafff5a7f58c853033d809f7e2

closed 2024-03-27 by Qwerty-OFF