rusefi/uaefi #59: certain UAEFI boards do not initialize 5VP below certain 12v input voltage when they should totally do so cause vehicle batteries can drain lower than that

ggurov, 2024-04-07:
Some uaefi boards (BAD BOARD) seem to not initialize 5VP regulator (GPIO::E10) when voltage is below 11.4 (ish, we ran some tests on one example board). This leads to a NO START condition. If board is connected to USB, 5VP initializes and everything is fine, even below no start threshold voltage. and power keeps on.

Other board examples (GOOD BOARD) will boot down to 6-7v input voltage just fine.

When this board (BAD BOARD) is powered below booting threshold, DFU mode still works, so MCU is alive and powered. 

Same board loaded with 407-disco firmware initializes just fine (which is odd, cause GPIO::E10 is something totally not power enable, it's assigned to something in the stepper territory, so may be it just so happens to work.

Anyway, please make UAEFI initialize 5VP more reliably.

rusefillc, 2024-04-07:
@ggurov any more details?

Which revision seems affected? What VBatt does TS show?

Do you have a bad board in NJ? What the minimal set of wires to reproduce the issue?

ggurov, 2024-04-07:
not sure about revision, we will check and record this.

minimal set of wires is 2-3:

1 ground
2 vign (if only this is hooked up, it raises the threshold, and board is worse at booting at lower voltages )
3 +12v main (forget what this is called) - if this is hooked up in addition to vign, board will init at lower voltage.

i do not have a power supply that can adjust below 11.6 or so, completely avoiding this. i may look into finding a 9v power brick to test my boards with.

when the board does not boot, COM port does not initialize, and TS will not connect. BUT DFU can be entered with buttons.

when the board does not boot, PE10 (en for U411 secondary regulator) stays low, when the board boots, that's at 3.3v.

when the board does not boot, 5VP led stays off.

ggurov, 2024-04-07:
both test boards are revA, one is bad, one is good

rusefillc, 2024-04-07:
@ggurov please confirm lack of capacitors next to SD as in https://github.com/andreika-git/hellen-one/issues/353

rusefillc, 2024-04-07:
See also #25

ggurov, 2024-04-07:
problem with looking at Vbat is usb connection will initialize the MCU making it turn 5VP on.

we did a test with a usb cable that does  not have power connected, good board would initialize and report to TS down to 4.4 volts, bad board would not initialize below 11.4 or so.

ggurov, 2024-04-07:
but the bad board could enter DFU via the same powerless usb cable below threshold

ggurov, 2024-04-09:
bad board, running disco firmware initializes.

however, if we assign E10 to lets say "main relay", and below about 12.6v turn that on, the board goes offline.

so something is messed up with 5VP regulator, since turning it on causes MCU to go offline.

rusefillc, 2024-04-11:
3v goes away, time to replace TLS115?

[image]

ggurov, 2024-04-11:
[image]

5v bridge to EN on tls115 external 5v regulator,  (5v to pin 7)

a fix has been proposed and tested for basic functionality
https://rusefi.com/forum/viewtopic.php?f=4&t=2748

rusefillc, 2025-10-16:
@ggurov what revision is your board?

rusefillc, 2025-10-16:
@ggurov @dron0gus good news I am able to reproduce
bad news https://github.com/rusefi/rusefi/commit/af428b608f2d7853c15d7aa955bf8743d4eb7122 did not help

dron0gus, 2025-10-17:
@rusefillc BOR settings are stored in options area of flash.
So it needs to be re-flashed to BOR_Level_None to be disabled.
But totally disabling BOR is rude. I'm trying to implement some kind of soft start. But no luck yet.

rusefillc, 2025-10-17:
@ggurov @dron0gus i think things have improved with https://github.com/rusefi/rusefi/commit/62d7dec3851909a81b85637dbb24b4776a75b61d

dron0gus, 2025-10-17:
Adding 470uF cap to V33 power rail solves hw part of issue.
3.3V no longer drops below BOR threshold. Even with threshold set to 2.5V with:
`BOR_Set(BOR_Level_2);`

[image]

rusefillc, 2025-10-21:
@dron0gus please help with https://github.com/rusefi/uaefi/issues/97

rusefillc, 2025-10-31:
fix documented at https://github.com/rusefi/uaefi?tab=readme-ov-file#bluetooth-power-issue-fix

closed 2025-10-31 by rusefillc