rusefi/uaefi #63: alternative/nicer connector + enclosure combo

rusefillc, 2024-04-26:
Molex is not ideal.

What better connector + enclosure combo shall we use?

As is I belive uaEFI has 68 pins

rusefillc, 2024-04-26:
[image]

stwillocks, 2024-05-08:
from Facebook post:

 90 pin enclosure from AliExpress: 90 Pin Way Car Driving Computer Board ECU Aluminum Box Housing PCB Socket Plug For Small Siemens Car AS90-1 211PC902S0008
 https://www.aliexpress.us/item/1005002745770669.html?spm=a2g0o.productlist.main.11.58fa2d64ryFtQh&algo_pvid=a90da2e4-e5b0-44fc-8b90-3a390b0d003a&algo_exp_id=a90da2e4-e5b0-44fc-8b90-3a390b0d003a-5&pdp_npi=4%40dis%21TRY%212109.93%211962.35%21%21%2161.62%2157.31%21%402101fb0a17151666301423218e13f6%2112000021944513011%21sea%21TR%21711978867%21&curPageLogUid=h3aGKjRVdMcd&utparam-url=scene%3Asearch%7Cquery_from%3A&gatewayAdapt=4itemAdapt
 
 
 Matching road car unit that I found is:
 
 S110140002 RENAULT CLIO EMS3132, 8200326391, 8200326380

It needs confirmation. The clio one is seems little bigger as enclosure.

rusefillc, 2024-08-14:
that's https://github.com/rusefi/rusefi/wiki/uaefi121 or maybe uaefi90 later

closed 2024-08-14 by rusefillc