rusefi/uaefi #66: how to i hook up logic analyzer to both VRs?

rusefillc, 2024-08-11:
We have amazing "cam" and "crank" vias on MM100 but looks like these are routed into random hall/button inputs?

Q1: where are the best spots to solder a wire for logic analyzer?
Q2: do we have space on the board to add vias?

[image]

rusefillc, 2024-08-11:
[image]

[image]

Qwerty-OFF, 2024-08-14:
[image]

closed 2024-08-14 by Qwerty-OFF