rusefi/uaefi #71: Hellen-one mega MCU input filter caps not matching

P4ulBr4ndt, 2025-02-18:
C202 is 10nF vs 200, 201 and 203 1nF is probably not intentional? Or maybe it is and I don't get it
[image]
When building hellen-one boards it gets like this into BoM and is manufactured like this

rusefillc, 2025-02-18:
@P4ulBr4ndt what does https://github.com/rusefi/uaefi/blob/main/bom_pullups_uaefi-d.csv have?

rusefillc, 2025-02-18:
@Qwerty-OFF is this a usability issue - shall we mention "see pullups replacement file for actual values" on all relevant pages?

P4ulBr4ndt, 2025-02-18:
As a user I can report bom_replace file feels more documented then pullups replacement file. 
Might want to add it to hellen-one docs more? 
Also for the defaults maybe its still nice to have them matching 🤔

Qwerty-OFF, 2025-04-30:
Only in mm-100 for now
[image]

closed 2025-04-30 by Qwerty-OFF